PRBonn/PIN_SLAM #1: Noisy Mesh

hungdche, 2024-02-03:
Thanks for the incredible open-source project! I was running PIN-SLAM with the provided run_demo yaml file and the 100-frame KITTI from the download script, and got a quite noisy mesh. Meshes from the paper and video look really smooth, so I'm wondering what should I do to address this? Thanks
[image]

YuePanEdward, 2024-02-04:
Thanks for your interest in our project.
For the quick sanity check, the mesh you showed seems to be as expected.
To achieve a better mesh quality, you may add the following lines to `run_demo.yaml` config file:
```
continual:
  pool_capacity: 2e7
optimizer:
  iters: 40 
  batch_size: 16384
``` 
This will let the mapper train for more iterations with a larger batch size, thus leading to better mesh quality.

During the visualization using the PIN-SLAM visualizer, you may press `B` to show the back face rendering, and press `Ctrl + 9` to switch to the normal direction mesh color. The marching cubes voxel resolution by default is 30cm here. To have a finer reconstruction, you can either press `[` to decrease the marching cubes resolution or define the desired resolution (for example 20 cm) in the config file by adding `mc_res_m: 0.2` to the end of the config file under `eval:` category.

This is the mesh I got with the above settings:

[image]

You can get even better reconstruction quality by tuning the parameters in the sampler, mapping, tracking, neuralpoints, and loss class. For the sanity test, we simply use the default settings. 

For better visualization, you may try Open3D offline visualizer for specific rendering settings of material, light, and shadows.

I hope this will help address your concerns.

hungdche, 2024-02-04:
Thank you for your response! I just have one more question. At the end of the run, how can I save the reconstructed mesh of the entire run? I add `save_mesh: True` at the end of `eval` in config, but it only save the mesh of the last frame.

YuePanEdward, 2024-02-04:
From my perspective, when you turn on the visualizer, the mesh of the entire run will be saved by default. If it's not the case, you may press `G` to toggle on the global map visualization.

If you are running at a server without X service and cannot turn on the visualizer, you may inspect and save the reconstructed mesh after the SLAM process by:
```
python vis_pin_map.py ./experiments/xxxx_name_of_the_run 0.2 neural_points.ply  mesh_20cm.ply
```

hungdche, 2024-02-04:
Thank you for your help! I will close this issue now. Just want to share this beautiful reconstruction from one of the Waymo segment:

[image]

closed 2024-02-04 by hungdche