PRBonn/PIN_SLAM #5: Difficult to operate at the sensor frequency in kitti 00

lhthq, 2024-03-06:
Hi~, thanks for your great work and the open source.  We tested on a RTX3080Ti GPU with 12 GB memory. The average consuming time per frame is 245.8ms in the whole KITTI 00 dataset. I think the performance of RTX3080Ti is similar to the A4000. The parameters are all default.  Can you give us some advice? Thank you!

YuePanEdward, 2024-03-09:
Thank you very much for your interest in our work.
You may try the parameter settings mentioned here: https://github.com/PRBonn/PIN_SLAM/issues/6, which conducts fewer iterations for mapping and registration. The reconstruction quality would deteriorate due to insufficient training but the localization accuracy would stay almost the same.

lhthq, 2024-03-09:
thank you. I try it. It takes about 121ms on single RTX3080ti GPU.

closed 2024-03-09 by YuePanEdward