PRBonn/PIN_SLAM #6: About time consumption: our Titan V  can not

zhSlamer, 2024-03-08:
Thank you for your wonderful work！

I run your code on **GPU Titan V , 12 GB** and  
set the pool_capacity: 5e6, batch_size:8192; 
But I don't know **why my time consumption is so long**.
On  Kitti-00, running time for 43 min.
I want to ask **how to adapt it to my GPU for real-time  operation**;

this my time details on Kitti-00:
[image]

sometime odometry time consumption over 200ms:
[image]

Thank you for your help!

YuePanEdward, 2024-03-09:
Thank you very much for your interest in our work.
The reported consuming time is similar to the operation time on my CPU.
You may first check if `device` is accidentally set to `"cpu"` (default is `"cuda"`).
If it's not the case, then you may try the following parameters:

```
process:
  min_range_m: 3.0
  max_range_m: 80.0
  min_z_m: -3.5
  vox_down_m: 0.12
sampler:
  surface_sample_range_m: 0.25 
  surface_sample_n: 2
  free_sample_begin_ratio: 0.5
  free_sample_end_dist_m: 1.0
  free_front_sample_n: 1
neuralpoints:
  voxel_size_m: 0.4
loss:
  sigma_sigmoid_m: 0.08
  loss_weight_on: True
  dist_weight_scale: 0.8
  ekional_loss_on: True
  weight_e: 0.5
  numerical_grad_on: True
continual:
  batch_size_new_sample: 1000
  pool_capacity: 1e7
  pool_filter_freq: 5
tracker:
  source_vox_down_m: 1.0
  iter_n: 20
  term_deg: 0.05
  term_m: 0.001
pgo:
  map_context: True
  pgo_freq_frame: 30
  context_cosdist: 0.2
  min_loop_travel_ratio: 4.0
optimizer: # mappers
  iters: 10 
  batch_size: 8192
  adaptive_iters: True
```
which conducts fewer iterations for mapping and registration. It takes about 80ms on single A4000 GPU.

zhSlamer, 2024-03-09:
Thank for your help!
It looks better than before.
I think i might have something to do with my GPU utilisation, which is always low.
I will keep looking into it.

closed 2024-03-10 by zhSlamer