PRBonn/PIN_SLAM #9: odometry evaluation

Zhangjyhhh, 2024-03-27:
Thanks for your excellent work!!!
I run this codes in kitti_04,and get absolute trajectory error 0.34  and your paper is 0.1 in table IV. So how can i get the same result?
[image]

YuePanEdward, 2024-03-27:
Thank you very much for your interest in our work.

Unfortunately, there is some randomness due to the random seed and the specific hardware used.
You may firstly check if you are using the original `run_kitti.yaml` configure file. 
If so, you may try other random seeds by adding `random_seed` under the `setting` section in the configure file.

Here are the [results](https://uni-bonn.sciebo.de/s/ySWvouod6PJsEz4) I got just now:

[image]

jbehley, 2024-03-28:
Please ensure that you use the right ground truth poses provided by KITTI.  They should be located in `dataset/poses/04.txt`, when you extract the original KITTI odometry files.

I accidentally used first the SemanticKITTI poses, which are generated with SuMa (called `poses.txt` inside the KITTI folder). With the SuMa poses, I also got a **0.36** ATE[m], which seems to be in the similar range as your numbers.

With the correct KITTI ground truth poses, my results on a RTX 2080 are **0.147** ATE[m].

Zhangjyhhh, 2024-03-29:
> Please ensure that you use the right ground truth poses provided by KITTI. They should be located in `dataset/poses/04.txt`, when you extract the original KITTI odometry files.
> 
> I accidentally used first the SemanticKITTI poses, which are generated with SuMa (called `poses.txt` inside the KITTI folder). With the SuMa poses, I also got a **0.36** ATE[m], which seems to be in the similar range as your numbers.
> 
> With the correct KITTI ground truth poses, my results on a RTX 2080 are **0.147** ATE[m].
> 
> Hope that helps. Can you check if this resolves your problem?

Sorry for that  i just don't realize the semanticKITTI poses are generated with SuMa. And i use the ground truth poses provided by KITTI , now i get the result is 0.128 ATE[m]. Your result is right and thanks for you remind me

closed 2024-03-29 by Zhangjyhhh

jbehley, 2024-03-29:
no worries. thanks for checking.