PRBonn/PIN_SLAM #13: issue about mesh

whwh747, 2024-04-13:
Thank you for your outstanding work. I have a question about the kitti00 dataset. I observed that there are some blank areas behind the car. Is this because the radar did not detect the point clouds at the rear?
[image]

whwh747, 2024-04-13:
I have followed your suggestion and adjusted the mes_res to 20cm.

YuePanEdward, 2024-04-14:
Thanks for your interest in our work.
Yes, those blank areas behind the car are mainly because there are no LiDAR observations over there due to occlusions.
Without using prior knowledge, it's impossible to have a reasonable guess of the shape of those regions.
However, you may decrease the size of the parameter `mesh_min_nn` under `eval` section. This parameter controls the minimum number of nearby neural points for a valid SDF prediction for mesh reconstruction. We can make use of the inpainting ability of MLP-based hybrid representation to have a smooth guess of the surface in the region without enough observation.  Additionally, you may try to increase the `map_surface_ratio` (default 0.5) under `neuralpoints` section to assign more neural points away from the surface.

whwh747, 2024-04-15:
Thank you very much for your timely response. I followed your suggestion to reduce the mesh_min_nn in config.py to 2, which significantly improved the ability to fill gaps. That's simply fantastic! Could you please explain why this value can affect the ability to fill gaps during the reconstruction process?
[image]

YuePanEdward, 2024-04-15:
Thanks for your kind words.

`mesh_min_nn` controls the minimum number of nearby neural points for a valid SDF prediction for mesh reconstruction. For a query position, if the number of neural points in its neighbourhood is smaller than `mesh_min_nn`, then this query position would be masked for marching cubes-based mesh reconstruction, thus leading to the holes in the back of those parked vehicles. Using `mesh_min_nn = 2` means that once we find more than two neural points close to the query position, we will trust the SDF prediction here and use this SDF value to build the mesh. Therefore, though we don't directly observe the region, the decoder will anyway output a continuous SDF prediction using the features of those two nearby neural points. Note that during online training, we are using 6 nearest neighbouring neural points. Using only two neural points for SDF prediction may generate unrealistic artifacts. 

Generally speaking, a smaller `mesh_min_nn` leads to higher completeness at the cost of lower accuracy. And vice versa.

whwh747, 2024-04-16:
I got it， thank you very much！I will close this issue.

closed 2024-04-16 by whwh747