PRBonn/PIN_SLAM #16: Add pure-localization mode

YuePanEdward, 2024-04-30:
It would be nice to have a pure-localization mode so that we can localize the robot inside a pre-built map. This may also work for the revisited region. We can get rid of the repeated mapping and only do the localization when there are not much new observations.

wendwosenbb, 2024-07-22:
@YuePanEdward  any updates on this?

YuePanEdward, 2024-07-28:
> @YuePanEdward any updates on this?

Thanks for your interest. I will update soon.

YuePanEdward, 2024-08-27:
The pose tracking version of localization with a loaded pre-built map has already been added. 
Therefore, this issue is closed temporarily. The documents will be updated. 
We can reopen it later for the global localization version.

closed 2024-08-27 by YuePanEdward

nathanredmond123, 2025-12-04:
@YuePanEdward I'd like to bump this thread.. You say in an earlier comment that pose-tracking version with a loaded pre-built map is already implemented.. And that documents will be updated. I do not see this in the documentation, nor do I see the code that implements this. Is this something that was never merged into main branch? This feature would be very useful for me.