PRBonn/PIN_SLAM #18: torch.cuda.OutOfMemoryError: CUDA out of memory

zfkdsg, 2024-05-04:
Hello, when I try to run your code run_demo.yaml，the following error has occurred:
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 396.00 MiB. GPU 0 has a total capacity of 3.81 GiB of which 348.00 MiB is free.
My gpu is RTX3050,How can I modify this program to make it run properly.
Thank you!

zfkdsg, 2024-05-04:
By the way，my GPU memory is 4GB

kevintsq, 2024-05-04:
Reducing the batch size in the config file may help.

zfkdsg, 2024-05-04:
thanks I will try it out。

YuePanEdward, 2024-05-04:
Additionally, you may decrease the size of the voxel hashing buffer: https://github.com/PRBonn/PIN_SLAM/blob/1c2fc2fcc910520e0aa9d7e8da2d889f456f0c4b/utils/config.py#L94
These would also take up `self.buffer_size*8` Bytes, which equals to 400 MB here.

closed 2024-05-06 by YuePanEdward