PRBonn/PIN_SLAM #20: How to understand the stability of neural points?

marrblue, 2024-05-23:
Hi @YuePanEdward 

Thanks for your excellent work and great open-sourced code!
I find that It is kind of difficult for me to understand the **stability** of neural points in the paper.
It seems that the larger the stability value, the less reliable the neural points are, and the easier nearby query points are to be eliminated.
But the stability value looks like just the sum of the weights given when the neural points are queried.
I am not sure my understanding is right.
How should we understand this stability intuitively?

YuePanEdward, 2024-05-23:
Thank you for your interest in our work and your kind words.

Intuitively speaking, we use stability to represent the (weighted) count of training at a certain position, reflecting the level of map fitting. During online mapping, our goal is to quickly adapt to newly observed regions, which have low stability due to limited training, rather than focusing on areas that have been extensively trained with previous observations. However, once an area has been thoroughly observed, positions with low stability (trained with only a few samples) might indicate the presence of dynamic objects or measurement outliers. Consequently, these neural points can be pruned from the global map.

You may visualize the stability of each neural point in PIN-SLAM visualizer by pressing `P` (optionally `G` to visualize the global map) and then pressing `/` three times. The darker the color, the higher the stability.

marrblue, 2024-05-24:
> Thank you for your interest in our work and your kind words.
>
> Intuitively speaking, we use stability to represent the (weighted) count of training at a certain position, reflecting the level of map [...]

Thanks for your easy-to-follow explanation！ This design is reasonable.
So in my understanding, dynamic objects -> less trained -> lower stability.
But I found that, both in the paper and the code, it seems like points with lower stability are regarded as static objects
while with higher stability are regarded as dynamic objects.   Is it a typo?
[image]
```python
        # Strategy 1 [used]
        # measurements at the certain freespace would be filtered
        # dynamic objects are those have the measurement in the certain freespace
        static_mask = (certainty < self.config.dynamic_certainty_thre) | (
            sdf_pred < self.config.dynamic_sdf_ratio_thre * self.config.voxel_size_m
        )
```

YuePanEdward, 2024-05-24:
Aha, the aforementioned map pruning is done for the neural points after the SLAM process. 
For this equation you mentioned, it is used for online dynamic filtering of the input point cloud. We filter those measured points who lie in the stable (stability > threshold) freespace (sdf > threshold). If you are certain somewhere is the freespace and now you get a new measurement here, it would be reasonable to say this new measurement belongs to some outliers or dynamic objects.
However, this does not always work when the robot is moving fast in one direction. For a more sophisticated moving object filtering in a offline batch processing fashion, you may refer to our recent work [4DNDF](https://github.com/PRBonn/4dNDF).

I hope this helps.

marrblue, 2024-05-25:
Thank you for your detailed explanation. Now I got it :) Good luck with getting this paper accepted!

closed 2024-05-25 by marrblue