PRBonn/PIN_SLAM #22: Z Drift in custom dataset

Xiangzhaohong, 2024-06-24:
Thanks a lot for this great work!

I meet some problems when I test pin_slma.

When I  test in my custom data, I found there will have large z drift (about 8~9m) in long term(5~6km) travel. and the loop detector will be failed.
I tried to adjust some parameters in pgo and loop detector but got no improve.

Is there any idea to solve this, I will try to count distance only use x,y and ignore z for loop detection, but I dont think this is a good solution.

thanks again.

YuePanEdward, 2024-06-24:
Thanks for your interest in our work.
Could you please provide us with some information such as the type of the LiDAR sensor. Is it a 360-degree spinning LiDAR or a front-facing Livox-like LiDAR? 
For the loop detection, you may try the following parameter settings:

```
pgo:
  map_context: True
  pgo_freq_frame: 30
  context_cosdist: 0.3
  virtual_side_count: 10
  local_loop_dist_thre: 20.0
```

which would search for more candidate scans for loop detection with long run and large odometry drift

Xiangzhaohong, 2024-06-24:
@YuePanEdward thanks for reply! 
my lidar is rslidar-128, I will try this params.
besides, your  local_loop_dist_thre is determined by this line:
`self.local_loop_dist_thre = config_args["pgo"].get("local_loop_dist_thre", self.voxel_size_m * 5.0)`

Xiangzhaohong, 2024-06-25:
Thanks, I will try to solve this problem through other sensors.
and close this issue.
thanks again for reply!

closed 2024-06-25 by Xiangzhaohong