PRBonn/PIN_SLAM #24: Transformation matrix

Mahdikhoorishandiz, 2024-07-07:
Hello,
Thanks for the great work and for open-sourcing it.
I want to use another method for registration. When I change the T_mat matrix in the Tracker.py file (line 666) and I use the transformation matrix of my own method, I don't get a proper output.
Can you tell me which other parts of the code I need to change so that I can apply my own transformation matrix?

YuePanEdward, 2024-07-07:
Thanks for your interest.
The transformation matrix you refer to is the estimated result. I do not fully understand what you mean. Can you elaborate a little bit more or post the error message? Thanks

Mahdikhoorishandiz, 2024-07-07:
Thank you very much for your quick response
I want to use a neural network for registration instead of using estimator methods. I want to give the transformation matrix I got from the neural network to your code, but I don't know exactly what parts of the code I should change.

YuePanEdward, 2024-07-07:
In such cases, you actually don't need the tracking module. You can change the used pose for mapping with the pose estimated by your neural network (you need to calculate the estimated pose of the current frame under the map coordinate system).  
https://github.com/PRBonn/PIN_SLAM/blob/85d62606c4e21be8c73e67cd02f9a4076c58a036/utils/mapper.py#L138

closed 2024-07-11 by YuePanEdward