PRBonn/PIN_SLAM #26: poor result on new college dataset.

Zhangjyhhh, 2024-07-24:


YuePanEdward, 2024-07-24:
Hi, thanks for your interest in our work.

Here are the [meshes](https://uni-bonn.sciebo.de/s/TeyiZZg8bNbaUAP) I got for the NCD02_Quad mapping experiment. This folder includes the meshes of PIN-SLAM and two baseline methods (NKSR with KISS-ICP poses and SLAMesh). Numbers of the other four baseline methods are reported by Nerf-LOAM.

PIN-SLAM's evaluation results are:
```
MAE_accuracy (m),MAE_completeness (m),Chamfer_L1 (m),Chamfer_L2 (m),Precision [Accuracy] (%),Recall [Completeness] (%),F-score (%),Spacing (m),Inlier_threshold (m),Outlier_truncation_acc (m),Outlier_truncation_com (m)
0.11552514390375901,0.15247310990752128,0.13399912690564014,0.22192155677537126,83.47852386868307,80.73416443483003,82.08341197526036,0.02,0.2,0.4,2.0
```

We generally increase the iteration numbers for mapping, and increase the training data pool capacity. This would not affect the pose estimation accuracy but would slightly improve the reconstruction quality.

Zhangjyhhh, 2024-07-26:
> Hi, thanks for your interest in our work.
>
> Here are the [meshes](https://uni-bonn.sciebo.de/s/TeyiZZg8bNbaUAP) I got for the NCD02_Quad mapping experiment. This folder includes the meshes of [...]

thanks for your quick reply

closed 2024-07-26 by Zhangjyhhh