PRBonn/PIN_SLAM #27: IndexError: 'neighb_idx' is out of bounds for dimension 0 when extracting mesh after completing SLAM run

asda-cmd, 2024-08-21:
Hi there, thanks for your work! I just encountered a problem when the algorithm extracted mesh when it was finishing.
I first extract .ply files from rosbag using rosbag2ply.py, and I give .ply directory to the run_ncd_128.yaml file. Then I start the program by running the following command:
`python pin_slam.py /home/user/PIN_SLAM/config/lidar_slam/run_ncd_128.yaml -smc`
When the SLAM is finished and started to extract mesh, it gives the following error shown in the image happens in this [line](https://github.com/PRBonn/PIN_SLAM/blob/c72f1b0ac9e8e4da50f032a81dfc645360a01785/model/neural_points.py#L888) 
[image]

I also tried to run it on the GPU, it also gives the same error. This error only comes up in some of the datasets (e.g. newer college dataset sequence 02 long experiment with the parameter of run_ncd.yaml)

Thanks!

YuePanEdward, 2024-08-21:
Thank you very much for reporting the bug.

I guess the cause of the bug is as follows: 

In `pin_slam.py`, the order of the following two lines of codes is incorrect:
```
neural_points.recreate_hash(None, None, False, False) # merge the final neural point map 
neural_points.prune_map(config.max_prune_certainty, 0) # prune uncertain points for the final output    
```
, which should be:

```
neural_points.prune_map(config.max_prune_certainty, 0) # prune uncertain points for the final output   
neural_points.recreate_hash(None, None, False, False) # merge the final neural point map 
```
If we prune some points after recreating the hash indexing, the hash indexing might be out of the bound during the global mesh reconstruction. 
I will push the updated version soon.

For the mesh visualization, you may also turn on the visualizer by adding the `-v` flag when running the code.

asda-cmd, 2024-08-21:
Thanks for your quick reply! Now the problem has been solved.

closed 2024-08-21 by asda-cmd