PRBonn/PIN_SLAM #28: MaiCity01

whwh747, 2024-09-05:
Thank you for open-sourcing such an excellent work. May I ask why I got such a result on the MaiCity01 sequence? I ran this sequence using `python3 pin_slam.py -i dataset_path -vsm`.

whwh747, 2024-09-05:
[image]

YuePanEdward, 2024-09-05:
Thanks for your interest in our work. 

The simulation of the poses in MaiCity does not obey the common kinematics of robots. The car moves 1m without acceleration at the beginning of the sequence, indicating the car's acceleration is 20g, which is unrealistic. PIN-SLAM relies on the simple uniform motion model for the kinematic initial guess. It would lose track in the beginning and corrupt the map with the default configuration.  

To cope with this issue, you can edit the following lines in the config file (in your case, the default one `config/lidar_slam/run.yaml`:
```
setting:
  deskew: False # MaiCity didn't simulate the motion distortion
  kitti_correct: True # provide a initial guess of motion along x-axis for the first frame
```
You can get the following result:

[image]

whwh747, 2024-09-05:
thank you very much! i got it!

closed 2024-09-05 by whwh747