PRBonn/PIN_SLAM #29: issue

whwh747, 2024-09-06:
Thank you for your excellent work. I would like to ask if the variable `mesh_freq_frame` means that the mesh is rebuilt every `mesh_freq_frame` frames? How does this differ from reading the dataset every `mesh_freq_frame` frames?

whwh747, 2024-09-06:
thank you，
I saw the range input command. I used run_ncd.yaml, added pc_path, and commented out pose_path. I encountered a track lost issue (I used range 0 1300 5).

YuePanEdward, 2024-09-06:
Yes, as I mentioned, PIN-SLAM would lose track when processing with large frame intervals.
There are two options:
1. Instead of loading data with skipped frames by setting `--range x xxx xx`, you may try to add in the config file:
```
optimizer: # mappers 
   mapping_freq_frame: 5
```
This will then let PIN-SLAM do the mapping (SDF fitting) once per `mapping_freq_frame`. In this case, the tracking results would be worse (with not up-to-date and map underfitting) but may not lose track.

2. Use the pure mapping mode (with known poses)
You can then specify the path to the referenced pose `pose_path` in the config file and comment out the tracking part.
An example would be like:
```
setting:
  name: "test_ncd_128_easy"
  output_root: "./experiments"
  pc_path: "./data/ncd_128/quad_e/ply"
  pose_path: "./data/ncd_128/quad_e/poses.txt" 
  deskew: True
process:
  ...
continual:
  ...
# tracker:
#  source_vox_down_m: 0.6
#  iter_n: 100
#  valid_nn_k: 5
pgo:
 ...
eval:
 ...
```
I hope this can help you resolve your problem.

whwh747, 2024-09-06:
yes，I would like to know how you obtained the results for "quad" in Table 10 of your paper. When I run the sequence, it consistently shows tracking failure. Am I missing something?

whwh747, 2024-09-06:
I see. Does the data in your table involve reconstruction for every frame?

YuePanEdward, 2024-09-06:
Yes, it involves the reconstruction using all the frames for SLAM.
Besides, if you are using the Quad subset of the neural college dataset (64 beam, in `*pcd` format), you need to turn off the `deskew` option in the config file. This is because the `*.pcd` format point cloud does not contain the pointwise timestamp.

whwh747, 2024-09-06:
Thank you. Allow me to express my highest respect for your work on SHINE, Puma, KISS-ICP, VDBFusion, and Pin-SLAM!

closed 2024-09-06 by whwh747

YuePanEdward, 2024-09-06:
Thanks for your kind words.