PRBonn/PIN_SLAM #30: eval issue

whwh747, 2024-09-07:
Thank you for your excellent work. I have a question: I ran the maicity01 sequence multiple times, but each time the resulting mesh and pose were exactly the same. What could be the reason for this?
[image]

whwh747, 2024-09-07:
i find same issue in ncd dataset，Could there be an issue with my settings? I believe the results shouldn't be the same each time I run it.

YuePanEdward, 2024-09-07:
We fix the random seed for each run (by default, 42). Therefore, the results would be most of the time the same. To set a different random seed, you can add the flag  —seed 2024 (or other numbers) upon running.

whwh747, 2024-09-07:
thank you！

发自我的iPhone

closed 2024-09-07 by YuePanEdward