PRBonn/PIN_SLAM #33: How to get the SDF gradient from the input points to the SDF predictiom?

jiang-junlong, 2024-09-21:
Some tensor operation is Non-differentiable in neural_points.py, how to make sure the stability of the way we get the gradient?
The input coordinates are first discretized, neighbor searched, features extracted, weights calculated, SDF predicted and fused. Strictly speaking, the analytical gradient obtained by pytorch should not be the gradient of the SDF prediction value to the input position.
输入坐标首先离散化、近邻搜索、取特征、算权重、SDF预测和融合，通过pytorch获取的解析梯度严格意义上来说应该不是SDF预测值对输入位置的梯度。

YuePanEdward, 2024-09-23:
Thanks for your interest in our work.
The voxel-based neighborhood searching is not differentiable. However, it only acts to index the surrounding neural point in the nearby voxels and is not involved in the computational graph. All the steps afterward are differentiable. The relative coordinate between the query point and each surrounding neural point (whose coordinate is a constant) is used for latent feature concatenation and inverse distance weighting, thus involved in the computational graph of the final SDF prediction. You may read the paper for more details.

jiang-junlong, 2024-09-24:
Thank your for detailed answer and take up your valuable time.
And later I will re-edit my question and make my question more clearly after my recently busy days.

I think the unstanding how to get SDF gradient is very important, which significantly effect the localization accuracy.Intutively I think the Sdf decoder and the feature fusion way is not natural but the neural network is very powerful so Pin slam can work very well.

I mean the scan-to-implicit map registration can provide a more refined way to estimate pose compared to point-to-plane residual, which is the meaning of neural sdf map.

Photo slam also uses the LM optimization to tracking. The common way that pin slam and phone slam use these classic optimization methoda instead of the neural network backpropagation
arise my thinking.Because the pose is only 6dof, different from the map geometry or other property, it doesn't have to updated by backpropagation due to its low dimension.

As you mentioned in pin slam paper and other papers, geometry loss is easy to optimize but color loss or      photometry loss is hard to optimize by the graph solver.  And    
My thinking for this question is stopped here. I think that why visual odometry such as ORB slam and Vins always recover the 3d point.

YuePanEdward, 2024-09-27:
Thanks for your insightful comment. Now it seems the discussion is on the research concept instead of technical details. Therefore, I'd like to close this issue but we can still have the discussion under this thread.

closed 2024-09-27 by YuePanEdward