PRBonn/PIN_SLAM #35: Consultation on IMU and GNSS integration in PIN-SLAM

ramiro999, 2024-10-23:
I understand that PIN-SLAM focuses on globally consistent mapping using implicit neural representations. However, I wonder if the combination with inertial sensors and GNSS could improve robustness in scenarios where LiDAR odometry is not sufficient, or in cases where the sensor records data from a drone and the drone performs rotational maneuvers.

YuePanEdward, 2024-10-24:
Thank you for your interest in our work.
I agree with your suggestion—integrating IMU and GNSS observations into the LiDAR SLAM pipeline would indeed enhance the absolute accuracy and robustness of the system. Adding IMU preintegration and GNSS factors to the factor graph of PIN-SLAM should be feasible and not too cubersome. While these features are not part of the current PIN-SLAM framework, they represent a promising direction for future development. We'd also welcome any pull requests contributing to GNSS-INS-LiDAR SLAM fusion.

ramiro999, 2024-10-25:
I understand, thanks for the answer, I have had an idea of how to integrate data IMU and GNSS, but I do not know how much it matches, first, I have reviewed the code, and I notice that in the “utils” folder there is a module called “point_cloud2.py” under that same logic I have created a module called “imu. py” that reads imu data from rosbags files with point cloud, imu and gnss topics, but now I don't know where to import that module, maybe in the main module ‘pin_slam.py’ for the reconstruction, or add it in each specific module where it calls ‘point_cloud2.py’.

YuePanEdward, 2024-11-01:
Sorry for the late reply. 
For loading the per-frame IMU and GNSS data, you may use this line: https://github.com/PRBonn/PIN_SLAM/blob/306c83d13ecace46a10e8d63ceadb8349c569034/dataset/slam_dataset.py#L239.
Then you can provide the IMU and GNSS data in the __getitem__() function of each data loader class. For the return `frame_data` dictionary, you may add additional keys such as `'imu'` and `'gnss'`. For the processing of the IMU and GNSS data, you may create a utility class such as `IMUPreintegrator` or `EKFProcessor` and then take the `cur_frame_imus` as the input.

closed 2024-11-04 by YuePanEdward

peterWon, 2025-01-10:
> I understand that PIN-SLAM focuses on globally consistent mapping using implicit neural representations. However, I wonder if the combination with inertial sensors and GNSS could improve robustness [...]

@ramiro999 Hi, we developed a tightly-coupled LIO system based on PIN_SLAM and evaluated the positive role of inertial sensor when mapping on a drone platform. Here is the preprint (https://arxiv.org/abs/2501.04263).