PRBonn/PIN_SLAM #37: Parameter configuration for NCD02_Quad

wangziyue-nubot, 2024-11-29:
Hi! Thank you for this very interesting and solid work! I would like to follow up on this thread to ask if you could share the configuration parameter related to the NCD02_Quad dataset. I believe it would be very helpful for benchmarking. Thanks!

YuePanEdward, 2024-12-04:
Sorry for the late reply. 
I've added the config file for NCD reconstruction experiments:

You can run it by:
```
python pin_slam.py ./config/lidar_slam/run_ncd_recon.yaml -vls
```

And generate the final mesh by:
```
python vis_pin_map.py ./experiments/test_ncd_recon_xxxx 0.2 neural_points.ply mesh_out_20cm_18.ply 18
```

The results might be slightly better than the numbers reported in the paper due to the recent updates of this repository.

wangziyue-nubot, 2024-12-05:
Thanks for your reply

closed 2024-12-05 by wangziyue-nubot

peterWon, 2025-01-10:
@wangziyue-nubot Hi, I wonder whether you reproduced the results using 'run_ncd_recon.yaml' on ncd_example. I still encountered tracking failures on my device.