PRBonn/PIN_SLAM #38: Failure on MulRan sequences

M2-TE, 2025-02-03:
Running PIN_SLAM on the MulRan DCC01 sequence seems to result in PIN_SLAM losing track as per the output message:
```
Using data loader for specific dataset or specific input data format
Available dataloaders: ['apollo', 'boreas', 'generic', 'helipr', 'kitti', 'kitti360', 'kitti_mot', 'kitti_raw', 'mcap', 'mulran', 'ncd', 'nclt', 'neuralrgbd', 'nuscenes', 'ouster', 'paris_luco', 'replica', 'rosbag', 'tum']
PIN-SLAM starts
  3%|███▎                                                                                                   | 175/5542 [00:25<12:45,  7.01it/s]
Odometry evaluation:
Absoulte Trajectory Error      (cm): 475.520
SLAM evaluation:
Absoulte Trajectory Error      (cm): 475.520
  3%|███▎                                                                                                   | 175/5542 [00:26<13:41,  6.53it/s]
Lose track for a long time, system failed
```
This run was launched via the provided dataset loader:
```bash
python3 pin_slam.py ./config/lidar_slam/run_mulran.yaml mulran DCC01 -d
```
with an edited `run_mulran.yaml` config to use DCC instead of Kaiser.

The trajectory errors range from 5 to 5000 cms at around 2-4% progress with no discernable pattern, but the error is reproducible with both the GPU (CUDA 12.4) and CPU implementations (using -c flag).

YuePanEdward, 2025-02-04:
Hi, Thanks for your interest in our work.
You are recommended to use the original 40cm voxel size (or 30cm). 5cm is a bit too small. Check Table XIV in the paper for an ablation study on the voxel hashing resolution setting. The reason for tracking failure is that we only do the neighbourhood searching in a 5x5x5 voxels region so the valid range for SDF mapping is only about 3x the voxel size, resulting in 15cm in your case. This would lead to a tracking failure when the movement difference is larger than 15cm in a frame.

If you want to have a finer mesh reconstruction resolution, you can set the marching cubes voxel size to 5cm by adding so in the config file:
```
eval: 
mc_res_m: 0.05
```

In this case, the neural point voxel hashing size can be left to be 0.3 or 0.4 m. For better fitting accuracy, you may try to increase other parameters such as `feature_dim`, `mlp_level` and `mlp_hidden_dim`.

If you want to have a smaller neural point resolution for other purposes, there's another solution: 
increase `num_nei_cells` (under `neuralpoints` entry) from 2 to 3 or even larger. In such a case, we will use a larger neighbourhood searching range (7x7x7 voxels in the case of num_nei_cells=3). However, the computational speed would also slow down dramatically.

M2-TE, 2025-02-06:
Thank you very much for the detailed response!
Adjusting `mc_res_m` separately from the neural voxel size work make the mesh finer, but still not capture more detail; though I will have to try out the separate parameters you suggested. A higher compute duration would not matter much for me.

As for the tracking failure itself, would it not be possible to supply an estimated pose (e.g. via IMU data or otherwise) to use as a starting point for the nearest neighbour search to help prevent tracking failures?

YuePanEdward, 2025-02-07:
You are right, It would be possible to use IMU or wheel odometry for a better initial guess than the currently used uniform motion model. We have not released this yet but you may check the KN-LIO paper mentioned here: https://github.com/PRBonn/PIN_SLAM/issues/35#issuecomment-2581634887

closed 2025-02-07 by M2-TE