PRBonn/PIN_SLAM #40: "new GUI" related issues

haoyanan2024, 2025-03-04:
Thank you for your excellent work!
I noticed that you added the new GUI, but when I run "python3 pin_slam.py./config/lidar_slam/ run_demo.yaml-vsm ", I can't visualize it and it's stuck. 

PIN-SLAM starts
Invalid MIT-MAGIC-COOKIE-1 keyFEngine (64 bits) created at 0x7f3e3d64a010 (threading is enabled)
FEngine resolved backend: OpenGL
Invalid MIT-MAGIC-COOKIE-1 keyFailed to open X display. (exiting).
  4%|██████▋                                                                                                                                                               | 4/100 [00:02<00:33,  2.86it/s]

When I switched -vsm to -sm, the program worked fine. Are there any bugs in the new GUI? Eager for your answer!

YuePanEdward, 2025-03-04:
Hi, thanks for reporting the bug. I guess this is due to some issues with the X service. 
You may remove the following line and try again.
https://github.com/PRBonn/PIN_SLAM/blob/d65358f550f53eb8ff2f5706555861f3106ad3f2/utils/tools.py#L62
If you are running the code on a server, you may check whether X11Forwarding is set to yes in the ssh config file.

haoyanan2024, 2025-03-05:
Thank you very much for your answer! I solved it.

closed 2025-03-05 by haoyanan2024