PRBonn/PIN_SLAM #47: docker lidar_slam/run.yaml InvalidSyntax

ManonCortial, 2025-04-01:
In the docker, when I run 

`/src/PIN_SLAM# python3 config/lidar_slam/run.yaml  pin_slam.py ouster -i /storage/pcap_name.pcap -vsmd`

I get the error
```
  File "/src/PIN_SLAM/config/lidar_slam/run.yaml", line 1
    setting:
            ^
SyntaxError: invalid syntax
```

I have not modified the file, when opening it, it looks correct, with linux line endings.
```
cat -ve config/lidar_slam/run.yaml 
setting:$
  name: "test_pin"$
  output_root: "./experiments"$
  deskew: True$
```

Any idea of what is wrong ?

YuePanEdward, 2025-04-03:
Thanks for your interest in our work. I guess the order of the command is wrong: 
Try:
```
 python3 pin_slam.py config/lidar_slam/run.yaml ouster -i /storage/pcap_name.pcap -vsmd
```
instead of 
```
 python3 config/lidar_slam/run.yaml pin_slam.py ouster -i /storage/pcap_name.pcap -vsmd
```

ManonCortial, 2025-04-03:
OMG, sorry to have bothered you with my stupid mistake.

however, I ran into anther problem : ouster-sdk is not in your requirements.txt
Be carefull though, because you use ouster.pcap, which has been removed in version 0.11, so be sure to require version <0.11. Alternately, you can update your dataloader/ouster.py to the one in KISS-ICP, which has been updated to the newer versions.

Thank you for your quick answer.

closed 2025-04-03 by ManonCortial

YuePanEdward, 2025-04-03:
Thanks for your suggestion. I will change the dataloader with the one used in KISS-ICP.