PRBonn/PIN_SLAM #50: Problem about color lost in own dataset

mark220620, 2025-04-25:
Thank for your excellent work.

Hi, I'm using my own dataset with PIN-SLAM and the input is a colorized point cloud in .pcd format. However, the output point cloud (e.g., mesh.ply) does not preserve the color information, as shown in the first image.
The original input point cloud definitely contains RGB colors, as shown in the second image.

Is there anything I should set in the config file or code to make sure the color is used and retained in the output?

Thank you!

[image]

[image]

YuePanEdward, 2025-04-25:
Thanks for your interest in our work.

You can try the following:

1. Add to the setting section of the config file:
```
setting:
  .....
  color_channel: 3
  color_map_on: True
```

2. Option A: Run the code without a specific dataloader `-d`:

Currently, only the `ply` files support color IO. 

Run:
```
python3 pin_slam.py ./config/lidar_slam/your_config.yaml -i /path/to/your/point/cloud/ply/folder -vsm
```

Option B: Write your own specific data loader, return the 6-channel rgb point cloud in `__getitem__`.

mark220620, 2025-04-28:
Thanks for your reply. 

I covert my point cloud files from .pcd to .ply, and add the setting section of the config file. The output have color information.