PRBonn/PIN_SLAM #51: Cur_Source_torch is an empty tensor

XuMian-xm, 2025-05-22:
/home/ssd1/xm/PIN_SLAM-main/pin_slam.py:252 in run_pin_slam                                                                                          │
│                                                                                                                                                      │
│   249 │   │                                                                                                                                          │
│   250 │   │   T1 = get_time()                                                                                                                        │
│   251 │   │                                                                                                                                          │
│ ❱ 252 │   │   valid_frame = dataset.preprocess_frame()                                                                                               │
│   253 │   │   if not valid_frame:                                                                                                                    │
│   254 │   │   │   dataset.processed_frame += 1                                                                                                       │
│   255 │   │   │   continue                                                                                                                           │
│                                                                                                                                                      │
│ ╭───────────────────────────────────────────────── locals ─────────────────────────────────────────────────╮                                         │
│ │                   argv = ['pin_slam.py', '-i', '../data/data/PCD', '-sm']                                │                                         │
│ │              color_mlp = None                                                                            │                                         │
│ │                 config = <utils.config.Config object at 0x7406f99a0100>                                  │                                         │
│ │            config_path = 'config/lidar_slam/run.yaml'                                                    │                                         │
│ │       context_pc_local = tensor([[ 17.9451, -43.3704, -66.3249],                                         │                                         │
│ │                          │   │   [ 18.2347, -44.1534, -65.7615],                                         │                                         │
│ │                          │   │   [ 17.5691, -43.7467, -66.1450],                                         │                                         │
│ │                          │   │   ...,                                                                    │                                         │
│ │                          │   │   [-72.7364, -22.1432,  27.2789],                                         │                                         │
│ │                          │   │   [-71.9503, -24.1693,  24.2264],                                         │                                         │
│ │                          │   │   [-71.9280, -24.6516,  24.2261]], device='cuda:0')                       │                                         │
│ │               cpu_only = False                                                                           │                                         │
│ │           cur_edge_cov = None                                                                            │                                         │
│ │                cur_fps = 0.6419446611043429                                                              │                                         │
│ │ cur_frame_process_time = array([0.00878501, 0.44962144, 0.0119741 , 0.19124055, 0.00441623])             │                                         │
│ │           cur_iter_num = 12                                                                              │                                         │
│ │               cur_mesh = None                                                                            │                                         │
│ │           cur_odom_cov = None                                                                            │                                         │
│ │         cur_pose_torch = tensor([[-2.9491e-01,  6.3031e-01,  7.1815e-01, -8.8355e+01],                   │                                         │
│ │                          │   │   [-6.0597e-01,  4.5774e-01, -6.5059e-01, -8.9578e+02],                   │                                         │
│ │                          │   │   [-7.3880e-01, -6.2705e-01,  2.4696e-01,  1.0429e+01],                   │                                         │
│ │                          │   │   [ 0.0000e+00,  0.0000e+00,  0.0000e+00,  1.0000e+00]], device='cuda:0', │                                         │
│ │                          │      dtype=torch.float64)                                                     │                                         │
│ │          cur_sdf_slice = None                                                                            │                                         │
│ │         data_loader_on = False                                                                           │                                         │
│ │                dataset = <dataset.slam_dataset.SLAMDataset object at 0x7406f99c0100>                     │                                         │
│ │           dataset_name = None                                                                            │                                         │
│ │                 deskew = False                                                                           │                                         │
│ │               frame_id = 2646                                                                            │                                         │
│ │            frame_range = None                                                                            │                                         │
│ │                geo_mlp = Decoder(                                                                        │                                         │
│ │                            (layers): ModuleList(                                                         │                                         │
│ │                          │   (0): Linear(in_features=11, out_features=64, bias=True)                     │                                         │
│ │                            )                                                                             │                                         │
│ │                            (lout): Linear(in_features=64, out_features=1, bias=True)                     │                                         │
│ │                          )                                                                               │                                         │
│ │              init_pose = array([[-0.75172589,  0.33781155, -0.56638463,  0.        ],                    │                                         │
│ │                          │      [-0.2213988 , -0.93826957, -0.26576831,  0.        ],                    │                                         │
│ │                          │      [-0.62120106, -0.07438804,  0.78011259,  0.        ],                    │                                         │
│ │                          │      [ 0.        ,  0.        ,  0.        ,  1.        ]])                   │                                         │
│ │             input_path = '../data/data/PCD'                                                              │                                         │
│ │             last_frame = 3498                                                                            │                                         │
│ │               lcd_npmc = <utils.loop_detector.NeuralPointMapContextManager object at 0x7406f99c0310>     │                                         │
│ │ local_map_context_loop = False                                                                           │                                         │
[... cut ...]
│ │    voxel_size = 0.6                                      │                                                                                         │
│ ╰──────────────────────────────────────────────────────────╯                                                                                         │
╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯
IndexError: min(): Expected reduction dim 0 to have non-zero size.

YuePanEdward, 2025-05-22:
Hi, thanks for your interest in our work. 
Could you please check if that pcd file is valid (containing points)?

YuePanEdward, 2025-05-23:
Great, glad that you solved it. I will add this at the next update.

closed 2025-05-23 by YuePanEdward